adamgallas/fpga_accelerator_yolov3tiny #13: 请教一下ps控制输入特征图的部分

Colouredseal, 2023-05-11:
您好，我想学习您的代码和控制逻辑，但是在控制数据传输的时候对于每次怎么传八个通道的特征图有些疑问
void Set_Next_IFM_ADDR_LEN(){
	if(iter_ofm_post!=ofm_batch-1||iter_div_post!=fm_div_cnt-1||iter_ifm_post!=ifm_batch-1){
		ifm_send_task_enable=1;
		if(fm_size>256)	ifm_addr_fmbase=iter_ifm_pre*fm_size;
		else ifm_addr_fmbase=iter_ifm_pre*256;
		if(fm_div_cnt==1){
			ifm_addr_offset=0;
			ifm_send_len=fm_size<<3;
		}
		else{
			if(iter_div_pre==fm_div_cnt-1){
				ifm_addr_offset=(fm_row-fm_res-2)*fm_col;
				ifm_send_len=((fm_res+2)*fm_col)<<3;
			}
			else{
				ifm_addr_offset=(iter_div_pre*fm_div)*fm_col;
				ifm_send_len=((fm_div+2)*fm_col)<<3;
			}
		}
		ifm_addr_send=ifm_ddr_base_addr+((ifm_addr_fmbase+ifm_addr_offset)<<3);
	}
	else{
		ifm_send_task_enable=0;
	}
	return;
}

这部分代码我理解下来是 每个批次只发送了第一张图的对应块的基地址 和对应的长度   对于同一个批次的其他七张图 我在代码里没有找到具体的传输  那个<<3指的是int8的8位数据长度吗？ 我是按照<<3是指int8的长度来理解这段代码  您能告诉我这部分具体的传输形式吗谢谢了

adamgallas, 2023-05-14:
特征图的memory layout是：Ci/8 * H * W * 8，<<3相当于是乘8，因为我用的是64位总线，一次性能传8个通道。

closed 2023-05-21 by adamgallas

lllkjh, 2023-11-02:
> 特征图的memory layout是：Ci/8 * H * W * 8，<<3相当于是乘8，因为我用的是64位总线，一次性能传8个通道。

请问下这里的Ci是指输入通道数吗