adamgallas/fpga_accelerator_yolov3tiny #14: zcu104移植问题

wgq18, 2023-05-24:
您好，我尝试将工程移植到zcu104上，对dout_0[0]的约束有疑问。您的约束是：
set_property PACKAGE_PIN B8 [get_ports {dout_0[0]}]
set_property IOSTANDARD LVCMOS12 [get_ports {dout_0[0]}]

请问B8引脚的作用是什么？zcu104上有相似功能的引脚吗？

谢谢您！

adamgallas, 2023-05-25:
这个引脚是对mipi摄像机做开关的。你用不着的话可以把它删掉。我记得zcu104是没有mipi摄像机接口的，只有usb摄像机，这意味着mipi ip核，伽马矫正那一条摄像头支路你可以直接去掉了，换成你zcu104支持的摄像机。

wgq18, 2023-05-25:
请问您的vitis版本是2021.1吗？我在2021.1版本中建立c工程编译，有大量重复定义的报错。
[image]
[image]
[image]

closed 2023-05-28 by adamgallas