adamgallas/fpga_accelerator_yolov3tiny #16: 量化模型生成问题

DragMou1204, 2023-06-05:
你好，在运行yolov3tiny_quant时里面是直接用load_quant_model加载的量化好的模型，请问一下这个量化好的模型（.pth）我该怎么得到，是通过generate_quant_model这个生成吗，生成过程中还出现了报错，能请您解答一下吗
[image]

adamgallas, 2023-06-06:
抱歉啊，量化这部分python代码其实我自己写得不好，也很久没维护了。但是量化是一个很成熟的领域了，pytorch现在对量化也有很完善的api了。我建议你自己去做网络的量化。难的部分不在量化，而是在加速器的设计上。

DragMou1204, 2023-06-06:
感谢你的回复

closed 2023-06-06 by adamgallas

1218liu, 2023-10-28:
> 你好，在运行yolov3tiny_quant时里面是直接用load_quant_model加载的量化好的模型，请问一下这个量化好的模型（.pth）我该怎么得到，是通过generate_quant_model这个生成吗，生成过程中还出现了报错，能请您解答一下吗 [image]

请问一下量化好的模型（.pth）您得到了吗？能解答一下吗