adamgallas/fpga_accelerator_yolov3tiny #25: 想要对其他的目标进行识别与分类需要修改源码的哪些部分

1218liu, 2023-10-08:
请问一下我已经用python实现了水果的检测与分类，训练推理与量化文件皆以跑通，并且也已经复现了您的源码，能够通过板子的摄像头进行头盔的识别，现在我想用您的源码来进行水果的检测与识别，请问需要修改您源码的哪些地方，才能够在板子上来实现这个功能，您能够详细地解答一下吗？

1218liu, 2023-10-10:
作者你好，我把我生成的bin文件放到sd卡里了，您的源码没有改，可是最后只能获取到图像并没有框，请问一下需要修改您源码的哪些地方才能识别出我想要识别的东西？

adamgallas, 2023-10-10:
恭喜！
要跑其它网络，主要需要改ps端调度的代码，accel.c，accel_data.c，main.c可能都要改。你可能要读懂ps端的这些代码，再根据你的需求改。你这个水果的检测分类是要画多个框吗？比较遗憾的是这个工程目前只支持画一个框。

1218liu, 2023-10-10:
好的感谢作者，我用的就是您的yolov3-tiny网络，我只是换了个水果数据集，后续如果还有问题再来找您

1218liu, 2023-10-10:
请问一下作者我并没有像您说的跑其他的网路，就用您的yolov3-tiny网络，只是数据集换成了水果数据集，最终用您的工程来实现水果的检测分类，您说要跑其它网络的话，主要需要改ps端调度的代码，accel.c，accel_data.c，main.c可能都要改；那我没有用其他的网络跑，也是改ps端调度部分的代码？还是说改ps端别的部分的代码呢，假如说是要改ps端别的部分的代码，那还是改accel.c，accel_data.c，main.c这几个文件吗？

adamgallas, 2023-10-11:
我建议是尽量读懂这几个.c代码，我也没有封装地很复杂，全都是面向过程的函数，函数的含义也写得比较清楚，只不过是没有注释。你耐心看一遍代码你就知道需要改什么了

closed 2023-10-12 by adamgallas