adamgallas/fpga_accelerator_yolov3tiny #26: 请问python工程中如图2所示的int8量化后的模型是如何得到的？

1218liu, 2023-10-28:
[image]
[image]

adamgallas, 2023-10-29:
你可以再仔细看看python整个工程。另外，量化是个很成熟的技术，到处都有成熟的教程，pytorch也有成熟的api，这个工程只是个很粗糙的量化的小demo。你完完全全可以多了解一下量化的相关知识

closed 2023-10-29 by adamgallas

DragMou1204, 2023-10-31:
[image]

1218liu, 2023-10-31:
> [image]

请问·一下出现这个错误是怎么回事啊，你能方便解答一下吗
[image]

1218liu, 2023-10-31:
[image]

DragMou1204, 2023-10-31:
qq941079882