adamgallas/fpga_accelerator_yolov3tiny #27: 推理运行时的几个问题

lllkjh, 2023-11-01:
麻烦几个问题请教下：
1. 在hls_preprocess模块中，有一个convert函数，应该是将像素都下采样到0-127的取值范围了，那是不是后续输入到加速器的数据都是0-127的而非0-255？是不是根据量化定的？
2. 在main.c的258-260行，Setup_Video_Resize()函数里的，我看对ifmInbuf里三个元素的地址分别进行了设置，个人理解这里的[0][1][2]三个数组对应的就是输入视频流的RGB三个通道吧？但是在accel_data.c里122行的ifm_ddr_base_addr_in_list里，貌似只用了ifmInBuf[1]这个数组的数据，那是不是意味着只将G通道的数据输入到加速器了呢？这样没问题吗？
3. 第一层的输入应该是RGB三通道的特征图，但在yolov3tiny_quant.py里看到存到配置bin文件里的第一层通道数是8（ifm_list的第一个元素），这里没搞懂为什么是8，不应该是3吗？

谢谢

adamgallas, 2023-11-06:
不好意思回复晚了。
第一个问题。“应该是将像素都下采样到0-127的取值范围了”你是怎么判断的？我在函数中是通过一个查表完成一个映射的。这个查表是将原始的rgb像素值映射成网络输入的范围的，这个范围是量化过程的input quant stub指定的。建议你了解一下量化。
第二个问题。并不是，这里的[1][2][3]不是指RGB，而是输入缓冲区，输出缓冲区，和临时缓冲区。
第三个问题。加速器在通道并行的粒度是8，如果输入通道小于8，会补零补成8通道

lllkjh, 2023-11-07:
感谢回复，问题1里写的下采样可能不太准确，我就是看到hls_preprocess模块里的IFM_QUANT_TABLE.h应该是等效于把0-255每个值都除以二再取整嘛，对吧？

另外，我这边鉴于手边没有摄像头，就是直接把图片数据读取到ddr然后输入进加速器进行处理的，feature map的memory layout就是按照之前帖子里你回复的Ci * H * W * 8来处理的，按理说同一张图片输入到模型，输出应该也是恒定的，但运行实际结果每次推理结果都有波动，且应该是没有识别出来，我用的就是工程里口罩识别的模型，输出结果如下，现在好像也没查到哪里有问题，不知道老兄是否有任何建议或者思路，谢谢。

`YOLOV3_TINY begin!

[Time elapsed is 72905 us]

xleft = 184 xright = 323 ytop = 46 ydown = 221
cat: 0        //输入图片为未带口罩人脸，这里结果却一直是0（0是戴口罩）
prob: 6     // 这里输出数值为原本prob变量 * 1000，即这里置信度仅0.几%
YOLOV3_TINY begin!

[Time elapsed is 72908 us]

xleft = 184 xright = 323 ytop = 46 ydown = 221
cat: 0
prob: 6
YOLOV3_TINY begin!

[Time elapsed is 72904 us]

xleft = 184 xright = 323 ytop = 46 ydown = 221
cat: 0
prob: 6`

adamgallas, 2023-11-07:
我的工程里好像有2个模型的，一个是头盔识别，一个是口罩识别，通过一个按键信号对这两个模型进行切换的。你看看在ps端怎么把这个按键切换的功能给去掉，固定为口罩识别模型再试试？如果每次推理结果有波动，是不是数据没有load成功，或者没有刷新cache？

lllkjh, 2023-11-07:
刷新cache是指每回load图片数据的时候刷新吗？还是指别的阶段要刷新呢，除了load数据的部分，其他代码都是沿用的你这个工程里的代码就是了，模型切换应该是没问题，因为调试的时候试过都没有进到切换模型的分支（PS: 下图是图片load进ddr后的排布）麻烦了，感谢
[image]

adamgallas, 2023-11-07:
每回load图片数据的时候flush一下就好了。那图像的预处理呢？图像的预处理是怎么做的，我原来的工程还有一个hls_preprocessing的ip，图像有过这个ip吗？还是说自己在ps端做了这个预处理？

adamgallas, 2023-11-08:
那确实似乎没有其他问题了。。不好意思我也不知道哪里可能出错了。2年前的工程了，我自己也没有很好维护

closed 2023-11-08 by adamgallas

foooal, 2024-03-10:
> 感谢回复，问题1里写的下采样可能不太准确，我就是看到hls_preprocess模块里的IFM_QUANT_TABLE.h应该是等效于把0-255每个值都除以二再取整嘛，对吧？
>
> 另外，我这边鉴于手边没有摄像头，就是直接把图片数据读取到ddr然后输入进加速器进行处理的，feature map的memory layout就是按照之前帖子里你回复的Ci * H * W * [...]

想问下您，您说的Ci * H * W * 8 是什么意思呢

输入特征有40个通道，截断为5个8通道的意思吗？
如果是这样的话， H * W * 8 是什么意思呢

adamgallas, 2024-03-21:
> 想问下您，您说的Ci * H * W * 8 是什么意思呢
>
> 输入特征有40个通道，截断为5个8通道的意思吗？ 如果是这样的话， H * W * 8 是什么意思呢

是的。相当于在输入通道进行了tiling。没错，40个通道会切成5份，变成5*H*W*8。H*W*8即数据在内存的排布顺序。8在最内层。另外，在已经close的issue里我是看不到的，你要提问题可以reopen这个issue，或者新开一个issue