adamgallas/fpga_accelerator_yolov3tiny #28: 检测水果时出现问题

1218liu, 2023-11-01:
复现您的源码后，将F开头的bin文件（口罩检测）改成自己生成的bin文件（也是F开头），头盔检测的bin文件没动，FCS.bin文件的标签种类为苹果，香蕉，橘子三种，可是始终识别不到橘子，在pycharm上跑推理的时候也能检测到橘子，可是上板子之后始终是将将橘子识别成苹果，请问一下作者能解答一下问题出在哪里吗？

[image]
[image]
[image]
[image]
[image]

adamgallas, 2023-11-06:
不好意思 不太想动脑了 2年前的工作了细节很多都不记得了

1218liu, 2023-11-06:
请问一下看到您的FCS.bin文件中是有三个标签的，复现您代码的时候检测口罩也是只出现过前两个标签，请问一下您的第三个标签是什么意思？你的这个第三个标签有出现过吗？因为我主要就是把您F开头的bin文件换成了我的，然后加了个我的FCS.bin文件，出现不了第三个标签我怀疑是不是您的这个代码只能检测到两个种类，想问一下您当时没有检测到过您的FCS.bin文件中第三个标签（incorr）？
[image]

adamgallas, 2023-11-07:
有的，可以检测多个种类的，但是我忘了要改啥了。我之前也曾经完整地跑过yolo里面所有的类别的，只不过后面改成了只有两三个类的口罩识别。事实上这个yolo分类头和加速器无关，是在ps端完成的，你只需要改ps端的代码就行了

closed 2023-11-08 by adamgallas