adamgallas/fpga_accelerator_yolov3tiny #32: yolov3-tiny换成yolov5s以提高检测精度和实现同时检测多个物体的可行性问题

1218liu, 2023-12-23:
您好作者，之前已经复现了您的工程，看到您在介绍中的这样一段话，
[image]
因此有了在此基础上换成yolov5并后续对v5进行改进来提高检测检测精度的想法，但是前两天又看到您在别人的问题下说这个工程不是通用加速器，不支持yolov5，所以想确认一下这个加速器是不是只适用于yolov3-tiny的推理加速吗？如果不能支持yolov5那我对于yolov3-tiny算法进行改进来提高检测精度，加速器支持改进后的yolov3-tiny吗？
还有一个关于同时检测多个物体画多个框的问题：这个工程目前只能画一个框，那我修改hls_rect_ip的代码能实现画多个框的功能吗？或者说修改哪些部分能实现画多个框的功能呢？

adamgallas, 2023-12-23:
基本上其实不太适合其它神经网络的推理。两年前我写这个reademe的时候低估了这个难度，我现在把readme改了吧。我不建议你自己在此基础上修改，因为不好改，存在很多问题，yolov5的cspblock存在多路的concat，这个加速器不支持concat，仅仅是一个小demo而已，后续我也没有维护。
支持改进后的yolov3tiny吗？其实目前我这个已经是简化版本的yolov3tiny了，我把yolov3tiny的那一路分支给去除掉了，因为我没实现concat。理论上可以在ps端做concat，但也挺麻烦，如果你执意要做，只能祝你好运了。
至于画框，软件画框用opencv调一下就好了，硬件上画框我一直没找到好的办法，hls_rect_ip也只是个权宜之计。
所以不建议你拿这个工程当做毕设，比赛，或者其他项目，只建议把这个工程用来学习。

starkerfirst, 2023-12-23:
您好，请问您在复现的过程中有没有遇到axi总线挂死的问题，我运行过程中跑到xil_out32这个函数以后整个cpu就会挂死，请问您是怎么解决这个问题的？谢谢！

adamgallas, 2023-12-23:
> 您好，请问您在复现的过程中有没有遇到axi总线挂死的问题，我运行过程中跑到xil_out32这个函数以后整个cpu就会挂死，请问您是怎么解决这个问题的？谢谢！

大概率是因为IP的寄存器空间没有配置好，导致Xil_Out32的寻址出错，没找到PL端的IP。

starkerfirst, 2023-12-23:
感谢您的回复，但是我的addr是直接引用xparameter.h里的宏定义，不会出现未定义的问题。但确实就是在axi访问这一步出现的卡死，我看汇编是这样。请问您有比较好的解决方法吗

adamgallas, 2023-12-24:
> 感谢您的回复，但是我的addr是直接引用xparameter.h里的宏定义，不会出现未定义的问题。但确实就是在axi访问这一步出现的卡死，我看汇编是这样。请问您有比较好的解决方法吗

如果地址空间没问题，那只剩下pl端实现axi lite的逻辑出问题了。但是我这部分是搬的vivado的官方例程的代码。。所以我也不知道为啥出错了。

closed 2023-12-26 by adamgallas

NightWingbat, 2024-05-08:
> 您好，请问您在复现的过程中有没有遇到axi总线挂死的问题，我运行过程中跑到xil_out32这个函数以后整个cpu就会挂死，请问您是怎么解决这个问题的？谢谢！

您好，请问之前遇到的AXI总线挂死的问题您解决了吗？我目前也遇到了这个问题，想要请教一下。