adamgallas/fpga_accelerator_yolov3tiny #36: 量化权重生产问题

foooal, 2024-03-21:
首先感谢大佬。
我在复显权重这里遇到一点问题。

```python
img_path='gao_mask.jpg'
img = cv2.imread(img_path)
img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)  # 将BGR转换为RGB
from torchvision import transforms

# 预处理图像
transform = transforms.Compose([
    transforms.ToPILImage(),
    transforms.Resize((640, 640)),  # 假设模型输入尺寸为640x640
    transforms.ToTensor(),
    transforms.Normalize([0.485, 0.456, 0.406], [0.229, 0.224, 0.225])  # 标准化
])
img_tensor = transform(img).unsqueeze(0)  # 添加批次维度

float_model,dataloader_iter=load_model_data('face_mask.yaml','yolov3tiny_facemask.pt',416,False)
generate_quant_model(float_model,dataloader_iter,'yolov3tiny_facemask_quant.pth')
quant_model=load_quant_model('yolov3tiny_facemask.pt','yolov3tiny_facemask_quant.pth')
quant_model_evaluate_show(img_tensor,quant_model)
yolov3tiny_infer_para_gen(quant_model,32,'F')
```
目录如下
 python_prj
 ├── images
 └── labels(代码中好像没遇到)
 └── yolov3tiny_quant.py

最后在quant_model_evaluate_show函数报错，这个函数应该是看下分割的权重能否正确检测。
Input shape must be (N, C, H, W)!
File "D:\yolo_work\fpga_accelerator_yolov3tiny\python_prj\models\common.py", line 45, in fuseforward
return self.act(self.conv(x))
File "D:\yolo_work\fpga_accelerator_yolov3tiny\python_prj\yolov3tiny_quant.py", line 80, in quant_model_detect
x=quant_model[1].model[ii](https://github.com/huanggeli/yolov3tiny-ZYNQ7000/issues/x)
File "D:\yolo_work\fpga_accelerator_yolov3tiny\python_prj\yolov3tiny_quant.py", line 97, in quant_model_evaluate_show
res=quant_model_detect(x,quant_model)
File "D:\yolo_work\fpga_accelerator_yolov3tiny\python_prj\yolov3tiny_quant.py", line 320, in
quant_model_evaluate_show(img_tensor,quant_model)
ValueError: Input shape must be (N, C, H, W)!
请问您有遇到这个问题吗

adamgallas, 2024-03-23:
似乎是预处理图像部分的代码没有符合特定的layout。python debug应该挺好debug的吧。另外2024年了，现在很多量化框架都很成熟了，可以尝试用更加完善的量化api，我这个python工程写的不好，仅供学习参考

foooal, 2024-03-24:
感谢作者，目前思路是大概把您的工程跑通，然后写一个yolo_v5_tiny和改进量化。目前fpga已经移植成功。看来您yolov3tiny_quant的示例
```
float_model,dataloader_iter=load_model_data('kaggle-facemask.yaml','yolov3tiny_facemask.pt',416,False)
quant_model=load_quant_model('yolov3tiny_facemask.pt','yolov3tiny_facemask_quant.pth')
yolov3tiny_infer_para_gen(quant_model,32)
```
想问下```load_quant_model```和```generate_quant_model```有什么区别呢，在示例中为什么是调用```load_quant_model```而不是```generate_quant_model```

adamgallas, 2024-03-25:
load是读取已经量化好的权重pt，generate是生成量化后的pt。我肯定是已经generate过了，才去load的

adamgallas, 2024-03-25:
然而我对这个加速器是否能跑yolov5保持悲观态度。因为yolov5存在很多路的concat，可能需要魔改很多地方，也不一定能成功

foooal, 2024-03-25:
> 然而我对这个加速器是否能跑yolov5保持悲观态度。因为yolov5存在很多路的concat，可能需要魔改很多地方，也不一定能成功
主要是学习一下思路

closed 2024-03-25 by foooal