adamgallas/fpga_accelerator_yolov3tiny #37: 权重分割错误

foooal, 2024-03-29:
作者你好，我用docker拉取了最新版的yolov3，生成权重并且在PC上检测图片没有问题。
相应的权重用您的函数量化分割。
```python
if __name__ == '__main__':
    float_model,dataloader_iter=load_model_data('face-mask-detection.yaml','last.pt',416,False)
    generate_quant_model(float_model,dataloader_iter,'yolov3tiny_facemask_quant.pth')
    quant_model=load_quant_model('last.pt','yolov3tiny_facemask_quant.pth')
    yolov3tiny_infer_para_gen(quant_model,32,'F')
```
新版yolo_v3量化做了一些函数名的变化，我也做了相应的改变。
函数yolov3tiny_infer_para_gen中有如下报错
[image]
```python
Exception has occurred: KeyError
'1.model.0.act.scale'
  File "D:\fpga\app\test.py", line 132, in generate_para_list
    ascale=quant_model.state_dict()['1.model.'+str(index)+'.act.scale'].item()
  File "D:\fpga\app\test.py", line 239, in yolov3tiny_infer_para_gen
    w,b,wscale,ascale,cscale,azp,czp=generate_para_list(quant_model,index)
  File "D:\fpga\app\test.py", line 314, in <module>
    yolov3tiny_infer_para_gen(quant_model,32,'F')
KeyError: '1.model.0.act.scale'
```

请问有什么解决思路吗

adamgallas, 2024-03-30:
这个函数的干的事情是，从state dict里面去索引对应的scale。你可以保存一下你量化后的yolov3的state dict，然后load进来，print一下这个state dict有哪些key，再更具这个key去更改一下代码

foooal, 2024-03-31:
我打印了这个这个state dict。确实没有`act.scale（缩放因子）act.zero_point（零点）。
我理解这个应该和模型没有关系，是在量化这里没有设置好，但我检查了量化这里，确实没有检查出来在哪里设置这两个参数
```python
def generate_quant_model(model,dataloader_iter,quant_model_name):
    quant = torch.quantization.QuantStub()
    dequant = torch.quantization.DeQuantStub()
    quant_model=nn.Sequential(quant,model,dequant)
    quant_model = quant_model.to('cpu')
    quant_model.qconfig = torch.quantization.default_qconfig
    quant_model=torch.quantization.prepare(quant_model, inplace=True)
    with torch.no_grad():
        for t in range(100):
            print('-',end='')
            data=next(dataloader_iter)
            img=data[0]
            x=img.float()/255.0
            x = quant_model(x)
    quant_model=torch.quantization.convert(quant_model, inplace=True)
    torch.save(quant_model.state_dict(),quant_model_name)
```

adamgallas, 2024-04-01:
上面这个quant的的代码应该是挺标准的post training quant的流程，但也是两三年前的流程了，我不清楚现在torch有没有做出大改变。scale和zero point不是我们自己设置的，是torch的量化过程中自动帮我们统计生成的。理应经过torch.quantization.convert之后，你去print一个conv层或者mlp层的weight，它的weight就不再是单纯的float32的浮点数了，而是一个quant tensor，这个quant tensor内会包含scale和zero point。上面的代码你逐行运行一下自己看看吧

closed 2024-04-02 by foooal