adamgallas/fpga_accelerator_yolov3tiny #38: 关于我对于您的accel.c文件的Detect函数的理解

1218liu, 2024-03-29:
作者您好，看了您的vitis代码我的理解是 通过Detect函数得到了硬件画框所需要的框框坐标与其他数据。对于Detect函数的具体理解我认为三个for循环相当与遍历摄像头检测到的一帧特征图，curr_p相当于每个像素点的一个概率值，再通过if(curr_p>max_p)对于每个像素点的概率值进行比较 最后取最大概率的那个像素点的各种数据(max_p,trec,irec,jrec)，下一个for循环我认为是取所有种类中概率最大的那个种类为检测到的物体类别。再通过一系列坐标转换等操作 最终得到硬件画框所需要的框框坐标与其他数据。后续我想实现硬件上画多个框的功能，目前我是将if(curr_p>max_p)改为了if(curr_p>固定值)，也就是只要curr_p满足curr_p>固定值 那么就都保存下来，这样来实现通过Detect函数得到画多组框所需要的多组数据。不知道我对于Detect函数的理解是否有误，以及修改成if(curr_p>固定值)来实现一帧图像能够得到多组框框数据的做法是否可行。
[image]

1218liu, 2024-03-29:
[image]

adamgallas, 2024-03-30:
在软件上保存画多组框所需要的多组数据当然是可行的。但是在你具体怎么根据这些数据画框呢？我的工程里负责画框的hls核只支持在每一帧画一个框，我目前还没想到画多个框的办法，除非你集成了opencv的库，在软件上实现画多个框

1218liu, 2024-03-30:
感谢作者的解答，后续在软件上画框的话还不知道要怎么实现，目前只能试一试改一下负责画框的hls核，看看能不能支持每一帧画多个框的吧

adamgallas, 2024-03-31:
一个可能的思路是，比如你要画n个框，就有2n条横线，2n条竖线，hls核在遍历整张图片的时候，要判断当前的行数和列数是否和2n条横线和2n条竖线的行数和列数相等。但这样加起来就会有4n个if的判断语句，而且要全部flatten，所以不知道能否满足时序要求

1218liu, 2024-03-31:
感谢作者提出的建议，后续我再研究一下，如果有疑问的地方再来请教您

closed 2024-03-31 by adamgallas