adamgallas/fpga_accelerator_yolov3tiny #41: vivado无法打开工程中的BD文件

NightWingbat, 2024-04-26:
我使用的是Vivado 2018.3版本，打开工程之后无法看到BD文件。提问之前翻看了一下以往的issue，说是采用2019.2往上的版本。请问有2018.3的解决办法吗？（在网上找了一些发现没什么效果，再装2019.2的版本比较耗时间）
[image]

NightWingbat, 2024-04-26:
具体的报错信息如下：（之前根据报错信息在网上一条条找过解决办法，也添加了Xilinx的MIPI IP）
[image]

adamgallas, 2024-04-26:
似乎是ip版本无法向下兼容的问题。可能没法避免更新vivado了

NightWingbat, 2024-04-26:
我现在更新到了2019.2。BD文件还是打不开，并出现了以下报错，而且显示无法找到cam_wrapper.v这个头文件。
[image]
[image]

NightWingbat, 2024-04-26:
我现在更新到了2019.2。BD文件还是打不开，并出现了以下报错，而且显示无法找到cam_wrapper.v这个头文件。
[image]
chong'xi
重新保存之后BD文件还是打不开，并且提示找不到cam_wrapper.v这个头文件。
[image]
[image]

NightWingbat, 2024-04-26:
我现在更新到了2019.2。BD文件还是打不开，并出现了以下报错。
[image]
重新保存之后还是打不开BD文件，并且显示找不到cam_wrapper.v这个头文件。
[image]
[image]

NightWingbat, 2024-04-26:
请问您是用的2020版本吗？能把版本号告诉我吗？我再试一下。或者您看可能是什么其他的原因？

adamgallas, 2024-04-27:
抱歉。。我的vivado工程的版本是2021.1，git clone下来你可以从vivado的vivado.log里看到log里面是2021.1。但是hls的工程是2019.2。但是理应打开个block design不会那么费劲，你可以试试按只读模式打开。用不同的版本打开vivado最好备份一下，如果这个工程之前被2018.3打开过（非read only模式），我不确定vivado会对工程更改了哪些东西。。。

NightWingbat, 2024-04-27:
抱歉打扰了。挺奇怪的。我都是按只读模式打开。但是依旧没法打开block design。我翻看了下以往的issue，不知道是不是board file的缘故，或者我升级到2021.1试试。

adamgallas, 2024-04-27:
也有可能是board file的原因吧。可以试试去digilent的官方github账号，把genesys zu3eg这个板子的board file下下来，放在vivado安装目录的放board file的地方

NightWingbat, 2024-04-27:
好的好的。我试一下。谢谢了。

look121212, 2024-04-28:
请问vitis项目版本也是2021.1的吗

adamgallas, 2024-04-28:
只有hls是2019.2，其它都是2021.1。因为vivado hls只到2019.2,2021.1的hls变成了vitis hls

look121212, 2024-04-28:
我是否可以用2021.1的版本直接打开你的vitis项目，并下板运行？我用2023版本的从vivado项目开始重新实现，在下板的时候，读取SD卡时，一直重复提示重复定义，这是否与版本有关？

adamgallas, 2024-04-28:
理应可以的。但是做跨板子的适配很麻烦，不好弄，牵一发动全身，除非你非常清楚各个模块是干啥的。2023能不能跑？我也不清楚，没试过，2021.1，genesys zu3eg这块板子，摄像头配好，是可以的。读取sd卡重复定义啥意思？你可以贴个报错，看看串口输出什么东西

look121212, 2024-04-28:
使用的就是gzu3eg的板子，输出图片在另一个新的issue里面

closed 2024-04-29 by adamgallas