adamgallas/fpga_accelerator_yolov3tiny #45: 生成sd 卡文件问题

look121212, 2024-05-01:
我使用yolo3tiny_quant 文件拆分权重时，提示我错误信息如下：
[image]
我更改了文件路径如下：
[image]
best .pt 文件是我训练的权重文件，.pth是我根据pt 使用torch 生成的，mask.ymal 文件内容如下：
[image]

adamgallas, 2024-05-02:
我这块代码写的不是很通用，如果你是自己训练的模型，会存在不对齐的情况。代码的原理是通过state dict中每个tensor的名字去索引的，如果你自己训练的模型和我原来的模型的state dict命名不一样，就没法对齐。你可能需要自己阅读一下具体的python代码，并且print一下你自己训练的网络的state dict的每个tensor的命名，自己手动改一下

look121212, 2024-05-02:
请问你的pth文件是怎么生成的，我查看了你的pt和pth文件，以及我的pt和pth文件，发现我的pth文件的state_dict和pt文件一样（这与你的并不一样）

adamgallas, 2024-05-03:
emm..这关系不大吧。报错是因为我的state dict的字段和你的state dict的字段不一样，你得手动改改代码。在yolov3tiny_quant.py下的generate_para_list(quant_model,index)函数里面改

closed 2024-05-06 by adamgallas