adamgallas/fpga_accelerator_yolov3tiny #46: 程序卡在了YOLOV3_TINY函数中的Inference函数

NightWingbat, 2024-05-08:
在PS端进行调试时，发现程序卡住了。经过Debug，发现程序卡在了YOLOV3_TINY函数中的Inference函数，当通过AXI_Lite总线给卷积核发送指令时出现了问题，尝试用ILA观察下AXI_Lite总线上的信号，但观察下来好像并没有发送读写的指令与数据，想要询问下解决办法。
[image]
[image]
[image]

NightWingbat, 2024-05-08:
[image]

adamgallas, 2024-05-09:
意思是ps端往pl端写寄存器的时候ila并没有观测到对应的axi lite总线有活动？只能怀疑地址空间没有分配对了。你得手动改一下ACCEL_CTRL_ADDR的定义，改成vivado address map里面的地址。

NightWingbat, 2024-05-09:
我之前检查过地址空间的问题。ACCEL_CTRL_ADDR的定义与PL端的地址空间是一致的。
[image]
[image]
[image]

adamgallas, 2024-05-09:
我也不知道为什么了。这里应该和加速器无关，ps端发起的写命令，axi_lite_waddr_valid肯定会拉高的。或者你的ila触发条件没弄对？

closed 2024-05-12 by adamgallas