adamgallas/fpga_accelerator_yolov3tiny #49: 想咨询一下数据流传输的问题

PRJ614, 2024-06-21:
1. 想问一下DDR中特征图数据的存放规则，是图1这样的吗，比如一张（224,224,3）的图像，是先按行再按列再到通道数吗，输出给pl也是按地址连续的发送过去吗，那8个ifm是按图2读的吗
[image]
[image]

2. pl端是否是按脉动阵列展开kernel并行再扩展到IFM,OFM通道并行？每个IFM与8个不同kernel相乘，8个不同的ifm并行与64个不同的kernel相乘

adamgallas, 2024-06-21:
1 tensor传输的维度是（Co/8, H, W, 8），dma每拍传64bit，一次性传8个byte，对应8个通道，先按行再按列
2 是的，没错

PRJ614, 2024-06-22:
`tensor传输的维度是（Co/8, H, W, 8）`这个没太明白，那数据流是按我画的这样传输的吗
[image]

adamgallas, 2024-06-22:
没错 是的

closed 2024-06-25 by adamgallas