adamgallas/fpga_accelerator_yolov3tiny #54: 关于替换网络架构的问题

BeamReflection, 2024-12-18:
作者您好，感谢您的project，非常详细很受用。

请问如果我要替换网络（把yolov3目标检测模型替换成图像压缩模型（也是卷积网络））请问差异还是很大吗？看到您在Q&A环节里提到不建议魔改，工程难度大，请问具体是个什么情况呢？
如果实在难以魔改的话，您能提供一下大致的思路吗？我现在已经完成了神经网络模型的训练与量化，之后的步骤就不知道怎么做了。

感谢您的回复。

adamgallas, 2024-12-18:
1. 首先确保网络有没有别的算子，目前只支持3*3卷积和2*2池化，最后的mlp层也还没支持，也还不支持resnet的shortcut
2. 这个工程实现不同尺寸的特征图的卷积比较死板，只能支持固定若干种成倍数的尺寸，目前加速器支持的特征图尺寸是和yolov3tiny的输入尺寸捆绑的，是416*256，你需要确保你的网络接受的输入也是这个尺寸，否则就得改pl端的代码。。这就很麻烦了
3. 以上都确保没问题后，得手动改accel.h, accel_data.h等驱动的代码，确保参数和网络对齐
4. 如果真的想弄，最方便的办法是，照着yolov3tiny的主干，重新训一个一模一样的结构的网络

BeamReflection, 2024-12-18:
> 1. 首先确保网络有没有别的算子，目前只支持3_3卷积和2_2池化，最后的mlp层也还没支持，也还不支持resnet的shortcut
> 2. 这个工程实现不同尺寸的特征图的卷积比较死板，只能支持固定若干种成倍数的尺寸，目前加速器支持的特征图尺寸是和yolov3tiny的输入尺寸捆绑的，是416*256，你需要确保你的网络接受的输入也是这个尺寸，否则就得改pl端的代码。。这就很麻烦了
> 3. 以上都确保没问题后，得手动改accel.h, accel_data.h等驱动的代码，确保参数和网络对齐
> 4. 如果真的想弄，最方便的办法是，照着yolov3tiny的主干，重新训一个一模一样的结构的网络

1. 网络应该只有3*3卷积和转置卷积，5*5卷积和转置卷积，还有relu。没有resnet。不知道您说的mlp层指的是什么？
2. 我的特征图尺寸的确不是416*256，这样的话需要修改很多很多代码吗？vivado和vitis部分吗？
3. 驱动等代码我会之后修改。
4. 您也看到第一点了，由于任务不同，网络架构也不同，我觉得很难照着yolov3_tiny的架构训练一个一样架构的网络。

adamgallas, 2024-12-18:
mlp指的是全连接。确实，如果需要支持其它的尺寸，具体需要改verilog里面的linebuffer模块，比较难搞。目前不支持5*5卷积。所以移植很困难

BeamReflection, 2024-12-18:
> mlp指的是全连接。确实，如果需要支持其它的尺寸，具体需要改verilog里面的linebuffer模块，比较难搞。目前不支持5*5卷积。所以移植很困难

已经了解到很困难了。起初也只是想借鉴这个整体设计思路，然后自己写来着。毕竟只有网络结构不同，整体设计流程应该是差不多的吧（存疑）。而且我用不到摄像头之类的外设，ps端的设计应该会更简单一些？

另外请问您能给个联系方式吗？（不方便的话也没关系）

adamgallas, 2024-12-18:
其实我还是建议用vitis ai的dpu的flow吧，自己写加速器还是比较麻烦。我微信是ADAM_GALLAS

BeamReflection, 2024-12-18:
> 其实我还是建议用vitis ai的dpu的flow吧，自己写加速器还是比较麻烦。我微信是ADAM_GALLAS

了解，非常感谢您！

closed 2024-12-18 by adamgallas