adamgallas/fpga_accelerator_yolov3tiny #55: 关于DSP资源问题

wanglaohan233, 2025-01-29:
作者您好！
本人新手，想要在ZYNQ7020的板上复现学习您的工程，发现DSP资源只有220，按照您readme的说法，可将乘法器映射为LUT，对此有些疑问
1.映射后的乘法器在运算周期上与高速DSP应该会有些差异，是否会影响到整体的加速过程？若强行周期对齐后是否会导致高速DSP在此次运算的速率降低？（因为向下兼容低速乘法器）
2.是否可以通过减少并行度来解决该问题？若是以220的DSP资源，220/9向下取整取24*2 = 48 ，可否可以将通道修改为6进6出并行策略来规避资源问题？（不清楚会不会影响到PS端的策略？）

期望您的回复，感谢！

adamgallas, 2025-02-02:
1 会的，7系列的zynq时钟频率大概率很难超过100MHz，而且可能LUT也会成为资源的瓶颈
2 理论上是可以的。但是整个加速器耦合度太高了，参数化没做好，牵一发动全身，如果让我自己改，我也会觉得很难改

wanglaohan233, 2025-02-02:
> 1 会的，7系列的zynq时钟频率大概率很难超过100MHz，而且可能LUT也会成为资源的瓶颈 2 理论上是可以的。但是整个加速器耦合度太高了，参数化没做好，牵一发动全身，如果让我自己改，我也会觉得很难改

感谢回复！
我尽量尝试一下修改
再次请教一下耦合度太高是指的PS侧部分的调度耦合吗？如果是的话您会建议我着重修改哪一方面的调度或者模块？

wanglaohan233, 2025-02-03:
> PS端和PL端很多的设计都和并行度绑定了。就比如说数据总线的位宽，是64，这就是一个和并行度绑定的设计。我建议还是参考学习为主，在这基础上改的话，其实还不如自己从头写一个，还能学到更多细节

明白了！谢谢佬的回答，我后续重构试试，祝您工作顺利！

closed 2025-02-03 by adamgallas